CANopenNode/CANopenNode #540: Issue with object 0x1011 (RESTORE parameters)

CARLOS226, 2024-09-10:
Hello everyone, I'm discovering the use of the CANopenNode stack in conjunction with a STM32L432 board.

The STM32L432 connected to a PEAK is working as expected, when trying to access to the OD objects by the CANOpen bus.
The OD objects can be saved on the EEPROM inside the STM32L432.

But I have an issue with the EEPROM management, and the signatures principle.

In fact, after any CANopen SDO write command to object 0x1011 (RESTORE parameters), the signatures are not good any more, and emergency message is sent on the bus.
To solve this issue today, I'm sending directly a second CANopen SDO write command to object 0x1010 (STORE parameters) in order to "repair" the signatures again on the EEPROM.

I would need to make this "repairing" behavior in the code of the stack itself, but I'm a bit lost on how to do it correctly.

Any help will be appreciate.

Thank everyone in advance.

CARLOS226, 2024-09-27:
to be erased

CANopenNode, 2024-10-07:
If you use heartbeat (which is recommended), node guarding must not be used.

If you have a reason to use node guarding instead of heartbeat, you have to enable the module, for example inside CO_driver_target.h: `#define CO_CONFIG_NODE_GUARDING (CO_CONFIG_NODE_GUARDING_SLAVE_ENABLE |                               CO_CONFIG_NODE_GUARDING_MASTER_ENABLE`

See also CO_Node_Guarding.h