CANopenNode/CANopenNode #541: HBstate is CO_HBconsumer_ACTIVE while target device is offline in function CO_HBconsumer_getNmtState.

Tom7296, 2024-09-13:
```C
int8_t CO_HBconsumer_getNmtState(
        CO_HBconsumer_t        *HBcons,
        uint8_t                 idx,
        CO_NMT_internalState_t *nmtState)
{
    CO_HBconsNode_t *monitoredNode;

    if (HBcons==NULL || nmtState==NULL || idx>=HBcons->numberOfMonitoredNodes) {
        return -1;
    }
    *nmtState = CO_NMT_INITIALIZING;

    monitoredNode = &HBcons->monitoredNodes[idx];

    if (monitoredNode->HBstate == CO_HBconsumer_ACTIVE) {
      *nmtState = monitoredNode->NMTstate;
      return 0;
    }
    return -1;
}

```

`monitoredNode->HBstate == CO_HBconsumer_ACTIVE` is true after I power off the target device, I get 0 return from CO_HBconsumer_getNmtState but not -1.

Is there another method to check if device online.

Tom7296, 2024-09-13:
Oh! The reason is canopenNode will not check HBstate until the first HB？But how to check device not on line.

trojanobelix, 2024-09-19:
his is exactly how it is defined in the Canopen standard, see CANopen Communication Profile (DS 301)