CANopenNode/CANopenNode #542: OD_RAM.x1001_errorRegister resets

Jan57555, 2024-09-24:
Hello.

I am using the CANopenNode stack on my B-G431B-ESC1. The basic communication works already.

Now I want to send information about errors to other devices. When setting a bit of OD_RAM.x1001_errorRegister in my program, it gets reset back to 0 again. Is there any reason for this? I don't have this problem with other OD-Objects.

Thanks for your help.

martinwag, 2024-09-25:
you need to use the functions provided by 301/CO_Emergency.h, e.g. [CO_error](https://github.com/CANopenNode/CANopenNode/blob/58012aae34928d823abed5a2556bdab91ed3f4a2/301/CO_Emergency.h#L431) and macros

Jan57555, 2024-09-25:
Thanks for the advice, I will check it out.

closed 2024-09-25 by Jan57555

Jan57555, 2024-10-23:
Hi. 
I am still having trouble using the error functions. Can somebody provide an example on how to use them correctly?
Thanks!

CANopenNode, 2024-10-25:
Please check https://github.com/CANopenNode/CANopenDemo for some tutorials.