CANopenNode/CANopenNode #544: incorrect data in OD entry

steff1982, 2024-09-25:
Hi,
I implemented a Canopennode on an STM32F302 as in the git tutorial, and so far it works.
I connect now to a Beckhoff PLC and start to communicate, using the EDS file of the OD Editor.
I created 1 TX and 1 RX PDO. Put a device type and also a heartbeat producer with 1000 ms.
everything works, but randomly I get wrong data in the OD entries!

e.g. from beckhoff I sen every 10 ms a uint8 like i=i+1; 
on the STM I have an OD_RAM.x6000_counter variable, which shows the variable from the PLC, but randomly I get the value 12!
also on other registers I get randomly the value 12. I checked everything, but I am not able to solve it.
I got error 8220 and 8240, but it is really random. 
I really need to trust the OD values. To get more information, I put a PeakCan on the line, and data send from PLC are correct.
Can someone help me or has someone the same issue?

martinwag, 2024-09-26:
try disabling can reception of this one rpdo and then place a memory breakpoint in your debugger. this should give you a stack trace where the write comes from.

steff1982, 2024-09-26:
Hi,

I am using the CANopennode the first time, so I do not know where I should disable the reception of the RXPDO. In the OD Editor and recompile?

 

Best regards,   

 

Stephan Haringer, PhD

CEO / Owner, Scientific-solutions S.r.l.

39020 – Castelbello / Ciardes

Italy (BZ)

Web:  <http://www.scientific-solutions.eu> www.scientific-solutions.eu   mail:  ***@***.***> ***@***.***

Phone: EU +39 347 680 5912

martinwag, 2024-09-26:
you can disable a pdo at runtime in the OD.