CANopenNode/CANopenNode #546: Compile error if CO_CONFIG_PDO_OD_IO_ACCESS is not enabled

trojanobelix, 2024-10-15:
Hi Janez,
no big deal, but I just came across it. So you might close it without feedback. 

actual master:

CO_TPDO_init uses OD_write_PDO_mapping to configure OD extensions.
OD_write_PDO_mapping is enabled with CO_CONFIG_PDO_OD_IO_ACCESS

If CO_CONFIG_PDO_OD_IO_ACCESS is not enabled  you got
_../CanOpenSTM32/CANopenNode/301/CO_PDO.c(1218): error: use of undeclared identifier 'OD_write_PDO_mapping'_

```
 /* Configure OD extensions */
#if (CO_CONFIG_PDO) & CO_CONFIG_FLAG_OD_DYNAMIC
(...)
    PDO->OD_mappingParam_extension.write = OD_write_PDO_mapping;
(...)
#endif
```

```
#if (CO_CONFIG_PDO) & CO_CONFIG_PDO_OD_IO_ACCESS
(...)

/*
 * Custom function for writing OD object "PDO mapping parameter"
 *
 * For more information see file CO_ODinterface.h, OD_IO_t.
 */
static ODR_t OD_write_PDO_mapping(OD_stream_t *stream, const void *buf,
                                  OD_size_t count, OD_size_t *countWritten)
{
(...)
}
#endif
```

trojanobelix, 2024-10-15:
Or in other words:
PDO->OD_mappingParam_extension.write  uses OD_write_PDO_mapping.
But OD_write_PDO_mapping does not exist without CO_CONFIG_PDO_OD_IO_ACCESS

CANopenNode, 2024-10-16:
That is strange. PDO.c contains:
https://github.com/CANopenNode/CANopenNode/blob/29c5b64131654b6f6f2171c64a1d5ce8031bcaa0/301/CO_PDO.c#L27-L31

There should be a preprocessor warning before compile error.

trojanobelix, 2024-10-17:
You get both errors:
`../CanOpenSTM32/CANopenNode/301/CO_PDO.c(34): error: Dynamic PDO mapping is not possible without CO_CONFIG_PDO_OD_IO_ACCESS
   34 |   #error Dynamic PDO mapping is not possible without CO_CONFIG_PDO_OD_IO_ACCESS
      |    ^
../CanOpenSTM32/CANopenNode/301/CO_PDO.c(1218): error: use of undeclared identifier 'OD_write_PDO_mapping'
 1218 |     PDO->OD_mappingParam_extension.write = OD_write_PDO_mapping;
      |                                            ^
2 errors generated.`

CANopenNode, 2024-10-18:
Yes, that makes sense. However, configuration has to be fixed in that case. I don't know, if it makes sense to change that part.

Those configuration bits may be frustrating from the beginning. If it is left unchanged, it is suitable for most cases. Add functionalities if necessary.

But it is also possible to make a minimum configuration. I managed to use the stack with old, small 16-bit microcontroller with 2kb of ram: https://github.com/CANopenNode/CANopenPIC/blob/master/dsPIC30F/CO_driver_target.h

trojanobelix, 2024-10-18:
> I don't know, if it makes sense to change that part.

Your code, your decision ;-)
I just want to let you know.

CANopenNode, 2024-10-18:
Actually I don't have better idea. Maybe,once in the future, we will find better solution for this complexity with configuration.

closed 2024-10-21 by trojanobelix