CANopenNode/CANopenNode #547: How Can I Modify Custom OD Value

BLU3ERRY, 2024-10-16:
I have created a custom Object Dictionary (OD) entry 0x2000 0 named InputVelocity on the master PC for motor control. My goal is to change this InputVelocity's value from the PC and control the motor via TPDO communication. I’m wondering whether it is better to access and modify this OD value using SDO communication or directly through a function. If it is possible to access and change the OD value directly using a function, how can this be done?

CANopenNode, 2024-10-16:
There are so many options...

If you are using Linux and canopend, you can use bash or python or anything via socket (SDO).
https://github.com/CANopenNode/CANopenLinux

(In Linux) you can also send raw can message with cansend from bash, which acts as PDO. cansend is part of can-utils for Linux.

Or you can use "canopen" library for python, which is another project, see there.

Or...

BLU3ERRY, 2024-10-16:
I’m sorry if my question sounds a bit unclear; I’m still new to computer-related topics. Thank you for your response. Are you saying it’s recommended to use SDO or CANsend for PDO-like communication when changing values? And is it correct that directly accessing it, like OD_RAM.x2000_InputVelocity = 12345;, isn’t advisable? Should I understand that even when the device modifies its own data values, communication protocols should still be used?

CANopenNode, 2024-10-17:
Yes, I don't understand the question clearly.
Of course, you can change your OD_RAM.x2000_InputVelocity directly. Then you just map it to TPDO and on the other device to RPDO ...

CANopenNode, 2024-10-17:
Now I saw your other issue. CANopen may be hard to start, but then it is very intuitive, once you understand basic principles. All you have to do is to correctly setup variables from object dictionary.
Variables can be accessed via SDO or PDO or SRDO(secure PDO). 

- SDO is usually used by operator, who wants simple access to everything. He also configures PDO mapping and communication parameters.
- PDO is used for automatic exchange of specific variables between devices automatically. Your approach with OD_RAM.x2000_InputVelocity is OK for that.

BLU3ERRY, 2024-10-17:
Okay I'll try it. Thanks for your helping

BLU3ERRY, 2024-10-18:
I have one more question.

When my motor (node 2) sends its position or velocity to the PC using TPDO, the communication parameter has a transmission type of 255 and an inhibit time of 1ms. It seems like the TPDO is only sent when the value changes (the event timer is not visible in the product manual's subindex).

When my PC (node 1) sends InputVelocity to the motor using TPDO, even if I set the communication parameter's transmission type to 254 or 255 with an event timer of 0 and an inhibit time of 1ms, it doesn't seem to send the data when the value changes. How can I resolve this?

If I set the transmission type to 255 with an event timer of 1, I can confirm that it sends continuously. However, with an event timer of 0, the PC only sends the TPDO once when it enters the operational state and does not send it afterward.

CANopenNode, 2024-10-18:
This is called "send on Change Of State". This functionality has to be specified by application. See https://github.com/CANopenNode/CANopenDemo/blob/40039f392b949123125a3f95c5cd5327fe31915e/demo/CO_application.c#L112-L129

BLU3ERRY, 2024-10-18:
I’m not sure if I should say this, but it feels like my year-long constipation has finally been resolved. Thank you so much

CANopenNode, 2024-10-18:
Glad to see that.