CANopenNode/CANopenNode #549: Implementing CO_LOCK macro when using CANopenNode with FreeRTOS

SergeTauger, 2024-10-17:
Hi,
I am searching a proper way to implement LOCK/UNLOCK macro when using CANopenNode with FreeRTOS. Among projects in readme there are two options:
1. [Use mutexes](https://github.com/martinwag/CANopenNode/blob/d8287cd25d687a9157dd712d95e7e756251e6fb5/stack/neuberger-FreeRTOS/CO_driver_target.h#L177)
A possible pitfail is that LOCKs can be called in interrupt context, so if we fail to acquire a mutex from interrupt we skip the guarded section
3. [Disable/enable interrupts globally ](https://github.com/CANopenNode/CanOpenSTM32/blob/ac995b1c773eb95632ee5afe64ac14691874e30e/CANopenNode_STM32/CO_driver_target.h#L149)
This option could possible interfere with native FreeRTOS portENABLE_INTERRUPTS especially when called from interrupt.
To solve the problem I've tried to implement LOCKs over FreeRTOS functions
```
/* (un)lock critical section in CO_CANsend() */
#define CO_LOCK_GENERIC(CAN_MODULE, field)									\
	do{																		\
		if(xPortIsInsideInterrupt() == pdTRUE){								\
			(CAN_MODULE)->field = taskENTER_CRITICAL_FROM_ISR();			\
		}else{																\
			taskENTER_CRITICAL();											\
		}																	\
	}while(0)
// Why disabling the whole Interrupt
#define CO_UNLOCK_GENERIC(CAN_MODULE, field)								\
	do{																		\
		if(xPortIsInsideInterrupt() == pdTRUE){								\
			taskEXIT_CRITICAL_FROM_ISR((CAN_MODULE)->field);				\
		}else{																\
			taskEXIT_CRITICAL();											\
		}																	\
	}while(0)
#define CO_LOCK_CAN_SEND(CAN_MODULE) 	CO_LOCK_GENERIC(CAN_MODULE, primask_send)

#define CO_UNLOCK_CAN_SEND(CAN_MODULE)	CO_UNLOCK_GENERIC(CAN_MODULE, primask_send)

/* (un)lock critical section in CO_errorReport() or CO_errorReset() */
#define CO_LOCK_EMCY(CAN_MODULE)    	CO_LOCK_GENERIC(CAN_MODULE, primask_emcy)
#define CO_UNLOCK_EMCY(CAN_MODULE) 		CO_UNLOCK_GENERIC(CAN_MODULE, primask_emcy)

/* (un)lock critical section when accessing Object Dictionary */
#define CO_LOCK_OD(CAN_MODULE)  		CO_LOCK_GENERIC(CAN_MODULE, primask_od)
#define CO_UNLOCK_OD(CAN_MODULE) 		CO_UNLOCK_GENERIC(CAN_MODULE, primask_od)

/* Synchronization between CAN receive and message processing threads. */
#define CO_MemoryBarrier()				portMEMORY_BARRIER()
```
but sometimes I end up in HardFault_Handler because of PendSV interrupt. So it seems I did something really wrong. Could you please point me to the proper direction?
Best,
Serge

martinwag, 2024-10-18:
I've implemented the FreeRTOS port. No stack code runs inside an ISR, so the non-ISR variants of the FreeRTOS functions can be used safely. We use FreeRtos Queues to transfer the CAN rx messages from ISR context to the thread context.

mlout, 2024-10-18:
Not sure if it is of any help. But we use a approach similar to yours @SergeTauger with critical sections, because we have interrupts with higher priority (that do not use FreeRTOS functions) that we do not want masked. You can view our implementation here: https://github.com/CANopenNode/CanOpenSTM32/pull/73/files#diff-be041ff98d9e03991ad2cba9902e3eabcd18ece7bb041e75da79cafc2ea8622d

> No stack code runs inside an ISR, so the non-ISR variants of the FreeRTOS functions can be used safely.
That does depend on where you call canopen_app_interrupt from. In our implementation we call it from a timer interrupt.

SergeTauger, 2024-10-29:
My bad: the mistake was in setting wrong stack size of deferred interrupt processing task (yes, the most frequent FreeRTOS problem). The code on locks works just as expected and with no errors. Thanks to everyone for help.

closed 2024-10-29 by SergeTauger