CANopenNode/CANopenNode #556: Dynamic OD Object Parameter Settings

eyupsayin, 2024-11-25:
Hi,

I am developing software with CANopenNode in CIA-417 profile. I want to use the "Object-6005 | Lock/Unlock Parameter" feature. Due to the locking function, I need to set the SDO access of the objects I specify at system startup to "no". Then, when the access password is detected correctly, I want to set the SDO access to "rw". However, in the OD.C file produced by the "EDSEditor.exe" application, a structure is created as static const. Therefore, I cannot make changes to the parameters.

**For example;**
* Initial => ODObjs[0].attribute = ODA_MB;
* After entering password =>  ODObjs[0].attribute = ODA_SDO_RW | ODA_MB;

**The structure I mentioned is on OD.c file line 261;**

    ```
    static CO_PROGMEM ODObjs_t ODObjs = {
      .o_1000_deviceType = {
          .dataOrig = &OD_PERSIST_COMM.x1000_deviceType,
          .attribute = ODA_SDO_R | ODA_MB,
          .dataLength = 4
      },
    };

Can you help me with this?

eyupsayin, 2024-12-04:
Hi,

I solved my problem without changing the "ODObjs_t" structure. I checked the "Object-6005 | Lock/Unlock Parameter" parameter using the Callback function. If it is locked, I created "SDO->state = CO_SDO_ST_ABORT" and abortcode. Thus, I solved the control of the parameter lock status within the Callback function and did not affect the CanOpenNode structure.

closed 2024-12-04 by eyupsayin

CANopenNode, 2024-12-04:
It seems a nice solution, thank you.
Sorry, I missed that issue.

eyupsayin, 2024-12-04:
I want to thank you for sharing such a valuable library. It is very valuable to me.
I wish you good work.