CANopenNode/CANopenNode #560: flagPDO usage in detecte data chage

lijinlong-yevs, 2024-12-26:
OD configs:
[image]
[image]
[image]
[image]
i see CANopenDemo :
`void app_programRt(CO_t *co, uint32_t timer1usDiff) {
    (void) co; (void) timer1usDiff; /* unused */

    /* Simulation: detect change of state of the variable and trigger TPDO, to
     * which variable is possibly mapped. In our example x2110_variableInt32[0]
     * variable will be mapped to TPDO. If program detects change-of-state of
     * its value, TPDO will be triggered for sending. (x2110_variableInt32[0]
     * variable can be changed by SDO.)
     */

    /* static variable is like global: initialized to 0, then keeps its value */
    static int32_t value_old = 0;
    int32_t value_current = OD_RAM.x2110_variableInt32[0];

    /* Detect change of state and trigger TPDO. Of course, variable must be
     * mapped to event driven TPDO for this to have effect. */
    if (value_current != value_old) {
        OD_requestTPDO(OD_ENTRY_H2110_variableInt32, 1); /* subindex is 1 */
    }

    value_old = value_current;
}`
when program is running, i modify OD_RAM.x2006_testPDO[0]'s value ,but PDO not send.
[image]
CANopenNode is work,but OD_requestTPDO func is not work,i think my od configs or flagPDO is wrong .Can tell me flagPDO usage ? please help me ,thanks thanks thanks

lijinlong-yevs, 2024-12-26:
[image]
[image]
add OD_extension_t in static OD_ATTR_OD OD_entry_t ODList, detect data chage can sned TPDO:
[image]
but i still dont know flagsPDO usage

CANopenNode, 2025-01-04:
flagsPDO should work only in CANopenNode v1.3(old version). It has no influence on CANopenNode v4, which has very different object dictionary principle.

lijinlong-yevs, 2025-01-06:
> flagsPDO should work only in CANopenNode v1.3(old version). It has no influence on CANopenNode v4, which has very different object dictionary principle.

Through the configuration object dictionary mapping, then initializing it and adding it to the custom extension definition, PDO will be sent automatically when a parameter change is detected.

In a more detailed context, this could refer to the following process in software development:

By setting up an object dictionary mapping through configuration, and then initializing this mapping before adding it to a custom extension's definition, the system is designed to monitor any changes to the parameters. When a change in these parameters is detected, a PDO (Process Data Object) message is automatically triggered and sent.

closed 2025-01-06 by lijinlong-yevs