CANopenNode/CANopenNode #573: CANopen cybersecurity new feature

nicolapatate, 2025-04-14:
Hello,

Any plan to add cybersecurity to CANOpen to comply with European Cyber Resilience Act. This will be required for new product on sale from December 2027.

Best Regards,
Nicolas

martinwag, 2025-04-15:
- CANopen currently doesn't have standards for implementing security
- CiA Canopen has a working group that discusses cybersecurity. See https://www.can-cia.org/cia-groups/technical-groups/technical-committee/ig06/ig06-sig02-hlp-cybersecurity

rusoku2, 2025-04-15:
Is this requirement necessary for local networks?

nicolapatate, 2025-04-15:
@martinwag thanks for the explanation

@rusoku2 
Yes, this is for any product with digital element which is connected physically or logically to any device or network. It includes not only HW but also SW product and Open Source Software. 

There are some exceptions for some markets like Automotive (cars, trucks, trailer), medical, marine...

Full application is December 2027 as part of CE marking.

ockeydockey, 2025-12-02:
To add to what has already been said here, Software Composition Analysis (SCA) will likely become quite a large part of complying with the multiple cybersecurity initiatives that will be going into effect in the coming years.  Without tagged releases it is very difficult to track and manage what version of this amazing library is being used in each project.

This especially hit hard when part of the API changed within the PDO system. It was difficult to reuse existing code that was written against the old API while trying to pull in the CANopenNode library into a new project.

On top of that, tools like BlackDuck SCA only scan tagged releases.

It would be really helpful to have more frequent tagged releases, even if they are just minor updates (`4.1`, `4.2`, etc.).

@CANopenNode, are there any plans on doing something like this in the future?

ockeydockey, 2025-12-02:
I just realized that the word `Releases` might mean multiple things on GitHub.  The tags are the important part.  I'm not requesting that you use the `Releases` feature of GitHub.

martinwag, 2025-12-03:
@ockeydockey i would second tagging versions more frequently. The last tag (v4.0) is 5 years old.

JuPrgn, 2025-12-03:
I think tags on the current and future versions are sufficient. No need to do any archaeology, for critical systems we prefer to use latest versions with all the fixes. If someone needs a tag for an older version this can be asked for this specific case.