CANopenNode/CANopenNode #574: Use physical value in PDO message callback return the raw value

gennartan, 2025-04-24:
I have an EDS file describing a VARIABLE with a Factor 1e-3.

When reading the physical value using SDO, the factor is properly applied.
When receiving the value through PDO and using the .phys attribute, I get the raw value.

Currently I have a workaround, I simply read the raw value and re-apply the conversion after reception of the PDO. 

Here is the EDS file.
```
... 

[3300]
ParameterName=Battery
ObjectType=0x9
SubNumber=4

[3300sub0]
ParameterName=Number of entries
ObjectType=0x7
DataType=0x0005
AccessType=ro
DefaultValue=4
PDOMapping=0

[3300sub1]
ParameterName=Vbat pin
ObjectType=0x7
DataType=0x0006
AccessType=ro
DefaultValue=0
PDOMapping=0
Factor=1e-3
Unit=V

...
```

CANopenNode, 2025-04-25:
I think, this is a question for https://github.com/canopen-python/canopen

gennartan, 2025-04-30:
Oh, you are right. 
Sorry for the mistake!