CANopenNode/CANopenNode #575: avoid dynamic memory allocation?

dakhnod, 2025-04-29:
As far as I understand, CANopenNode (specifically OD.c) currently uses quite a few `malloc` calls in order to save the object dictionary.

This causes Problems, since

1. The heap if often times too small and needs to be enlargened
2. Some platforms don't really dynamic memory allocation
3. The RAM usage cannot be estimated at compile-time (I think)

Is there a way to use CANopenNode without any dynamic memory allocation?

If I understand correctly, all the size data is available compile-time or even at code generation time.

This may also be an Issue of the EDSEditor...

Thanks for your work and cheers!

CANopenNode, 2025-05-07:
See https://github.com/CANopenNode/CANopenNode/blob/f454a3d82dca058cc64e441fbaefa68a391e4757/CANopen.h#L164-L171

You can define it, for example, in your CO_driver_target.h

All variables are created in CANopen.c

dakhnod, 2025-05-07:
Ah, great, thanks!

dakhnod, 2025-05-07:
Sorry, just for my understanding: Why doesn't that work with multiple ODs?

closed 2025-05-07 by dakhnod