CANopenNode/CANopenNode #586: CANopenNode's use of CiA 301 0x1010 and 0x1011 entries

Eqqm4n, 2025-07-17:
Hello all-
Apologies for having this here as an "Issue" but this project doesn't seem to have a "Discussions" tab.  The setup is Windows 11 using a Kvaser Leaf 3 to communicate with a Nucleo-G01BRE board with the CANopen stack through Zephyr.

The goal is to have the mutable data for some of the object dictionary (OD) entries stored in flash and not RAM.  The OD editing tools allow for specification of storage groups, but even though they have names like "ROM" or "EEPROM" (and these names have since been replaced by "PERSIST_COMMS" or the like) these names are completely meaningless on their own, that is, they could be "A", "B", or "C" for all CANopen cares.  The docs for CANopenNode say:

> CANopen provides OD objects 0x1010 and 0x1011 for control of storing and restoring data. Data source is usually a group of variables inside object dictionary, but it is not limited to OD.
>
> When object dictionary is generated (OD.h and OD.c files), OD variables are grouped into structures according to 'Storage group' parameter.

So I'm having three problems here.  Firstly, in the OD editing tools I'm not finding any way to link the storage groups available to categories that OD entry 0x1010 will care about.  For example, the same doc says regarding 0x1010:

> Sub index 2: Save communication parameters, UNSIGNED32

How then does anything in the OD source files get flagged as a "communication parameter"?

Secondly, how do these default parameters get spawned in ROM?  If OD 0x1010/11 are "storing and restoring data", then to where?  When I look at the memory map spawned after compiling, all the mutable data values are placed in RAM and there is no "default" copy in the non-volatile memory (NVM) to be used by 0x1010/11.  So how do they get used properly then?

Thirdly, it seems like 0x1010/11 isn't what should be used anyway for some use cases.  If I want to do a firmware upload, I need to make a Domain-class OD entry, but this can never exist in the RAM space to begin with as the entire firmware image would never fit in there.  So this entry can only ever exist in the NVM and in that case 0x1010/11 aren't the right tools to work with that anyway.

A colleague is working on the same problem butt with a different processor from TI.  In his project, he uses a callback somewhere in the CANopen RX call stack to check if the OD domain entry representing file uploads is being accessed, then redirects the data to a flash manager tool instead of letting CANopen do the write.  But this seems like a "hack" and not the proper way to manage this process.  Or am I wrong and this is in fact how to deal with this type of issue?

ks1930, 2025-08-15:
@Eqqm4n were you able to find these parameter defintions / classifications in OD ? i dont see where it is defined

Eqqm4n, 2025-08-16:
@ks1930 They're not.  It's been very confusing working with this setup because to quote Inigo Montoya "you keep using that word, I do not think it means what you think it means".  As outlined in my OP, there are references to "communication parameters", and in the object dictionary editors you might see storage group labels like RAM, ROM, PERSIST_COMMS, etc.  Well, forget all of these names and terms because they are completely meaningless and only serve to confuse the uninitiated.  For the purposes of this discussion, CANopen has _storage classes_ that I'll just refer to as STORAGE_X.  In the object dictionary editing tools, you'll have some default STORAGE_X types you can pick from (like RAM, ROM, etc.) or you can make as many new ones as you like.  Remember that the names only have meaning to you, the CANopen stack doesn't care.

So, let's say that I have my OD entries I intend for read-write use assigned to STORAGE_1 and all my read-only entries assigned to STORAGE_2.  In the exported c/h files both STORAGE_1 and STORAGE_2 will be assigned their own `struct` and instantiated in the RAM.  If my goal was to have STORAGE_2 really be stored in the flash, for example, I would have to go back and manually edit the files so that STOARGE_2 is now a `const struct`.

When an OD write is done to x1010:2, it's up to _your_ code to trap this write and handle the saving of the STORAGE_2 structure to NVM, if STORAGE_2 is what _you_ are considering the "communication parameters" of your OD.  Note again, that since the exported c/h files only support creating storage structures in RAM, you have to jump through hoops to get things linked to NVM.  For example, you either have to edit the c/h files to make the NVM entries `const`, or you could duplicate these entries into a different file where you flag them as the `const` versions.  Then you could require users of your device to have to do a write to 0x1011 to load in these values or copy them in automatically yourself on device boot.

CANopen allows for registering callbacks to OD accesses so you can insert your code in that manages all this stuff.  At the time I'm writing this, I haven't done it myself, but other colleagues have.  It is definitely a requirement is you need to support file uploads over CAN.